Koheron/koheron-sdk #540: Koheron 250-4 data acquisition tutorial?

aksingh-dev, 2021-07-28:
I have checked the board with given SD card (Ubuntu OS) . 
The board is working well. I connected it with PC via Serial port and Ethernet.

Now, I want to capture 4-channel ADC data , and transfer it to my PC using Ethernet.
I am looking for some tutorial for data acquisition!

Also, what is the length of maximum continuous data that can be captured? For all 4 ADC channels sampling at 250MSPS. 
Thanks in advance.

jeanminet, 2021-08-06:
You can have a look at this example:
https://github.com/Koheron/koheron-sdk/tree/master/examples/alpha250-4/adc-bram

Data is stored in FPGA BRAM which allows only 32768 samples to be acquired for each of the 4 channels simultaneously.

Acquiring more samples would require transferring the data directly to the RAM using DMA (Direct Memory Access).
An example of such DMA is available for the ALPHA250 (not the ALPHA250-4): https://github.com/Koheron/koheron-sdk/tree/master/examples/alpha250/adc-dac-dma

aksingh-dev, 2021-08-09:
Thank you for the reply,
While running the example "adc-bram", it shows ADC size = 65536 (_per channel??_) 

_In case of ADC_DMA: Do I only need to modify these 4 files for ALPHA250-4?_ 

Similar example is available here: https://github.com/rsarwar87/koheron-sdk/tree/master/examples/alpha250-4/adc-dma-origin
However, I am getting Errors for command >>    make CONFIG=examples/alpha250-4/adc-dma-origin/config.yml 
[error_output.txt](https://github.com/Koheron/koheron-sdk/files/6951858/error_output.txt)

[image]

rsarwar87, 2021-09-03:
you are mixing and matching different branches which is why you are setting that problem. ps_control registers are not available on all branches i believe.

aksingh-dev, 2021-09-08:
Hello Rashed,

Yes, you are right.. I was merging two different branches. 
However, I tried to debug and generated the instrument file (zip). While running it on the board, it causes error.

[image]

So, I restarted again with different setup! I am using the PC for other boards, so I can't re-install or change OS.
At present, I am using your branch with Vivado 2019.1 on Ubuntu 20.04.
Now, I am getting another error:
[image]

Could you upload Alpha250-4 ADC-DMA design example files (zip)?

Thanks.

-AKSINGH