Koheron/koheron-sdk #542: /usr/bin/arm-linux-gnueabihf-g++-8  complier error

13121186, 2021-10-06:
When I use:
/usr/bin/arm-linux-gnueabihf-g++-8
I got this compilation error:
cc1plus: error: switch -mcpu=cortex-a9 conflicts with -march=armv7-a switch [-Werror]

I had no such issue when I used 
/usr/bin/arm-linux-gnueabihf-g++-5

How to fix it?
Thanks.

tvanderbruggen, 2021-10-07:
Can you perform the following change in `os/os.mk`:

https://github.com/Koheron/koheron-sdk/pull/543/files

Also I just compiled it on Ubuntu 20.04 with g++-9 and I don't have your error.
Can you post a log of the error ?

13121186, 2021-10-08:
with your above suggestions, now I can compile the files without errors. However, browser brings out nothing, blank web page
When I use Ubuntu16.06 with g++-5, the FFT plot is successfully showing up on the browser.
Any further advices on this?
Thanks.

13121186, 2021-10-18:
Thanks.
==Check Ubuntu version:
Running: cat /etc/os-release
1. Inside Alpha250
Welcome to Ubuntu 16.04.7 LTS (GNU/Linux 4.9.0-xilinx armv7l)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
root@koheron:~# cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.7 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.7 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

2.inside WSL distro (Ubuntu-20.04)

cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.1 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.1 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

==check GLIBC
strings /lib32/libm.so.6 | grep GLIBC 
GLIBC_2.0
GLIBC_2.1
GLIBC_2.2
GLIBC_2.4
GLIBC_2.15
GLIBC_2.18
GLIBC_2.23
GLIBC_2.24
GLIBC_2.25
GLIBC_2.26
GLIBC_2.27
GLIBC_2.28
GLIBC_2.29
GLIBC_2.31
GLIBC_PRIVATE
GLIBC_2.1.3

13121186, 2021-10-19:
I think the OS image in the Alpha250 board should be upgraded to ubuntu 20.04 as well to match the WSL ubuntu20.04 where I build the Koheron image.
I tried downgrading my WSL ubuntu 20.04 to ubuntu 16.04 which matches the version in the Alpha250 board. Now it is working without issues.

nbhebert, 2022-08-04:
I am having a similar problem, although I am not using WSL to build instruments. I am running a native Ubuntu20.04 machine where I installed gnueabihf-g++-9 as chosen by default in the makefile. Do I understand correctly that the solution is to build instruments using Ubuntu16.04 for now? Some more info on my particular situation below, including error logs. Thank you!

1. One of the errors in the instrument compilation log is: 

/usr/bin/arm-linux-gnueabihf-g++-9 -flto -c -Wall -Werror -Wextra -Wpedantic -Wfloat-equal -Wunused-macros -Wcast-qual -Wuseless-cast -Wlogical-op -Wdouble-promotion -Wformat -Wmissing-include-dirs -Wundef -Wcast-align -Wpacked -Wredundant-decls -Wvarargs -Wvector-operation-performance -Wswitch-default -Wuninitialized -Wshadow -Wzero-as-null-pointer-constant -Wmissing-declarations -Itmp/examples/alpha250/adc-dac-bram//server -I./server/core -I. -I. -I./server/context -I./server/drivers -Iexamples/alpha250/adc-dac-bram/ -DKOHERON_VERSION=0.22.2.e3a9f77 -MMD -MP -O3  -mcpu=cortex-a9 -mfpu=vfpv3-d16 -mvectorize-with-neon-quad -mfloat-abi=hard -std=c++17 -pthread -lstdc++ -lstdc++fs -static-libstdc++ -o tmp/examples/alpha250/adc-dac-bram//server/i2c_dev.o server/context/i2c_dev.cpp

2. The instrument still compiles and after uploading it to the Alpha250, the koheron-server throws these errors:

Feb 11 17:03:13 koheron serverd[1861]: /tmp/live-instrument/serverd: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/live-instrument/serverd)
Feb 11 17:03:13 koheron serverd[1861]: /tmp/live-instrument/serverd: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/live-instrument/serverd)

3. Info on my OS and GLIBC :

cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04 (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

strings /lib32/libm.so.6 | grep GLIBC
GLIBC_2.0
GLIBC_2.1
GLIBC_2.2
GLIBC_2.4
GLIBC_2.15
GLIBC_2.18
GLIBC_2.23
GLIBC_2.24
GLIBC_2.25
GLIBC_2.26
GLIBC_2.27
GLIBC_2.28
GLIBC_2.29
GLIBC_2.31
GLIBC_2.32
GLIBC_2.35
GLIBC_PRIVATE
GLIBC_2.1.3

nbhebert, 2022-09-06:
Just an update that after reverting to using Ubuntu 20.04 to build instruments (as suggested here https://www.koheron.com/software-development-kit/documentation/setup-development-machine ) everything works fine.