Koheron/koheron-sdk #545: How to run a basic instrument on alpha 250

v-shetty, 2021-11-02:
Hi Team,

its a newbie issue.

i could able to setup and install all the prerequisites needed for the alpha 250 and i am able to build and run successfully.

but i have trouble in testing how the instrument running

I would like to know if i want to test the  loopback and fft instrument, how i need to connect the wiring.
do i have to use the RF ADC/DAC or Precission ADC/DAC
and also please let me know how to i connect them.

tvanderbruggen, 2021-11-02:
Hi,

Both instruments loopback and fft are using the RF converters.

To test the fft instrument, you can connect a RF DAC output (OUT0 for example) to a RF ADC input (IN0 for example).
Adjusting the OUT0 DDS frequency on the instrument web page you should see a peak at the same frequency.

To test the loopback instrument, you can connect a signal from a waveform generator (amplitude below 1 Vpp) to IN0 and look at OUT0 on a scope. You should observe the waveform generator signal replicated on the output (see for axample the *Input to output latency* section on the [ALPHA250 web page](https://www.koheron.com/fpga/alpha250-signal-acquisition-generation)).

v-shetty, 2021-11-03:
Hi Vanderbruggen,

Thank you for the response.

To test loop back, i did exactly what you suggested and i run the instrument. 

[tmp/examples/alpha250/loopback//loopback.zip] OK
curl -v -F loopback.zip=@tmp/examples/alpha250/loopback//loopback.zip http://192.168.1.100/api/instruments/upload
*   Trying 192.168.1.100...
* TCP_NODELAY set
* Connected to 192.168.1.100 (192.168.1.100) port 80 (#0)
> POST /api/instruments/upload HTTP/1.1
> Host: 192.168.1.100
> User-Agent: curl/7.58.0
> Accept: */*
> Content-Length: 737503
> Content-Type: multipart/form-data; boundary=------------------------17daabcbdaf300f5
> Expect: 100-continue
> 
< HTTP/1.1 100 Continue
< HTTP/1.1 200 OK
< Server: nginx/1.10.3 (Ubuntu)
< Date: Thu, 11 Feb 2016 16:28:45 GMT
< Content-Type: text/html; charset=utf-8
< Content-Length: 33
< Connection: keep-alive
< 
* Connection #0 to host 192.168.1.100 left intact
Instrument loopback.zip uploaded.curl http://192.168.1.100/api/instruments/run/loopback
Instrument loopback.zip successfully installed

Even after successfully running, i still dont see any signal at OUT0.
i give square wave signal 10Hz 0.5Vpp at IN0

Do you suggest me any debug ideas?

tvanderbruggen, 2021-11-03:
If you go on the setting page (192.168.1.100/koheron) is the loopback instrument status set to running ?

Also 10 Hz is a very low frequency, can you use something like 1 MHz ?

v-shetty, 2021-11-03:
I tried with 1Mhz , but still no output. in the webpage , status is Running for loopback.

Also , I tried the FFT intruments , it works totally fine, with web page i can control

tvanderbruggen, 2021-11-03:
In the loopback design (https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/loopback/block_design.tcl) we've commented out the 2 lines connecting ADC0 to DAC0 and ADC1 to DAC1.

Can you uncomment the 2 lines and rebuild the instrument ?

v-shetty, 2021-11-03:
@tvanderbruggen 

super , works good.
i appreciate your support
Thanks a lot

v-shetty, 2021-11-03:
closing the issue as it is resolved

closed 2021-11-03 by v-shetty