Koheron/koheron-sdk #546: How to sychronise ADC sampling with a trigger

v-shetty, 2021-11-08:
Hi All,

I would like to know, how can i start receiving ADC data based on the trigger for fixed number of sample?
Trigger is internal software trigger, i would like to start sampling from ADC, when DDS from BRAM is streaming to DAC

my approach (software solution) : using the given example ADC-DAC BRAM instrument 

writing python thread script , which listen to the trigger(software trigger) and capture the ADC data

Do you have any hardware solution?or software solution is good enough?

tvanderbruggen, 2021-11-08:
DAC data in BRAM are played synchronously with ADC data BRAM filling.
So in that sense ADC data are already triggered with DAC data output.

But maybe I misunderstood your problem ?

v-shetty, 2021-11-11:
@tvanderbruggen 
thank you.
One more question,
how to change the DAC and ADC size?
currently it is 65536 in adc-dac-bram example

tvanderbruggen, 2021-11-15:
You can set the size in the `memory` section of the [instrument configuration file](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/adc-dac-bram/config.yml).

The size is defined in the `range` section of each memory.