Koheron/koheron-sdk #551: can Koheron server used on any ARM LINUX?

v-shetty, 2022-03-16:
Hi @tvanderbruggen 

I have embedded device with ARM linux, i want give user a web page as user interface to control all the pins in the board.
my board can handle tcp program,

can koheron web server can be used on other boards?
https://github.com/Koheron/koheron-server

tvanderbruggen, 2022-03-16:
Hi,

Yes in principle you can build the server with drivers for other arm boards.

Now the server is maintained and integrated within the SDK for Zynq boards but the [server section](https://github.com/Koheron/koheron-sdk/tree/master/server) is relatively independent.
So you would have to isolate the server build part (mostly keep the [server.mk](https://github.com/Koheron/koheron-sdk/blob/master/server/server.mk) file).
Also you need to adapt the [context](https://github.com/Koheron/koheron-sdk/blob/master/server/context/context.hpp) to your board (you probably don't need the fpga_manger, zynq_fclk and the memory_manager).

Hope this helps ...

rsarwar87, 2022-03-16:
Here is a port to rpi3/4 to control an artix7 fpga over spi

https://github.com/rsarwar87/CmodA7-SkyTracker/tree/device-CmodA7/koheron-server