Koheron/koheron-sdk #554: Proper way to set Vivado run strategy

jahampel, 2022-05-13:
I'm trying to build the FPGA design with a different run strategy to potentially fix some small timing violations (in the neighborhood of 100 ps setup slack) that are left over after timing closure. So far I've found two way of doing this.

- Open the project through `make CONFIG=<path> open_project`, set a run strategy in the GUI and run synthesis and implementation from there.
- Put the Tcl command `set_property STRATEGY <run strategy> [get_runs impl_1]` in the `block_design.tcl` file so it will be evaluated when building the block design. 

The first one is, of course, not very well-automated and while the second one works just fine, changing the run strategy in a file that semantically should just be the block design doesn't seem to be the most elegant solution to me.

So my question is, is there an intended way of changing the run strategy (or better yet, run arbitrary Vivado Tcl commands) other than calling these commands in the `block_design.tcl` file? Like I wrote, this works perfectly fine and Vivado does utilize the right strategy then, it just looks a bit like a hack to me.

rsarwar87, 2022-05-13:
that is the way to do it. but i have that set_property command on my 
https://github.com/rsarwar87/koheron-sdk/blob/master/fpga/vivado/project.tcl#L36

add custom defination there and pass the $mode as an argument.

jahampel, 2022-05-13:
Thank you for the swift response; I haven't seen this script before. So I can just write `set mode "production"` in my block design, or define a custom strategy in the `project.tcl`. Perfect, this is exactly what I've been looking for. Thanks a lot!

closed 2022-05-13 by jahampel