Koheron/koheron-sdk #556: Python API: socket connection refused for all instruments other than 'fft'

bamiesking, 2022-11-25:
### Python API connection refused
I have an ALPHA250 configured on a local network with a static IP address, and I'm able to run `test_clock.py` successfully. However, when I try to run `test.py` with the `adc-dac-dma` instrument:

```Traceback (most recent call last):
  File "/home/ben/.local/lib/python3.10/site-packages/koheron/koheron.py", line 244, in __init__
    self.sock.connect((host, port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ben/dev/koheron-sdk/examples/alpha250/adc-dac-dma/test.py", line 51, in <module>
    driver = AdcDacDma(connect(host, name='adc-dac-dma'))
  File "/home/ben/.local/lib/python3.10/site-packages/koheron/koheron.py", line 54, in connect
    client = KoheronClient(host)
  File "/home/ben/.local/lib/python3.10/site-packages/koheron/koheron.py", line 247, in __init__
    raise ConnectionError('Failed to connect to {}:{} : {}'.format(host, port, e))
requests.exceptions.ConnectionError: Failed to connect to 169.254.1.110:36000 : [Errno 111] Connection refused
```

This is both with the other example instruments, and my own instrument. Both `text_clock.py` and the `test.py` are unchanged other than the hostname.

### FPGA not being reprogrammed
I don't know whether this is related, but it also seems that the FPGA fabric is not being reprogrammed when a new instrument is loaded, due to the following sequence of events:
1. Load `fft` and set OUT0 to some frequency; corresponding DDS output is measured on an external scope.
2. `make ... run` a different instrument (e.g. `loopback` or a custom instrument which drives adc_dac/dac0 in the block design)
3. Observe sine wave output from OUT0, unchanged from before the FPGA was supposedly reprogrammed.

### Other context
* I am working on Ubuntu 22.04 running in VirtualBox (Windows 10), with Vivado 2017.2.
* I have made [one small modification](https://github.com/bamiesking/koheron-sdk/commit/53041b206206866fe3d3ed1b40a9f99138beb9d8) to the `fpga.mk` file, to directly reference Vivado rather than looking for `settings64.sh`, which was not generated during the Vivado install for me. 

Please let me know if you need any more info; any help greatly appreciated!

bamiesking, 2022-11-25:
I quickly tested the `adc-dac-dma` `test.py` script on Windows 10, with both Python 3.7 and 3.11 to rule out a VirtualBox issue, and got a slightly different connection error:

```
Traceback (most recent call last):
  File "C:\Users\bak517\AppData\Local\Programs\Python\Python37\lib\site-packages\koheron\koheron.py", line 244, in __init__
    self.sock.connect((host, port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:/Users/bak517/development/koheron-sdk/examples/alpha250/adc-dac-dma/test.py", line 51, in <module>
    client = connect(host, name='adc-dac-dma')
  File "C:\Users\bak517\AppData\Local\Programs\Python\Python37\lib\site-packages\koheron\koheron.py", line 54, in connect
    client = KoheronClient(host)
  File "C:\Users\bak517\AppData\Local\Programs\Python\Python37\lib\site-packages\koheron\koheron.py", line 247, in __init__
    raise ConnectionError('Failed to connect to {}:{} : {}'.format(host, port, e))
requests.exceptions.ConnectionError: Failed to connect to 169.254.1.110:36000 : [WinError 10061] No connection could be made because the target machine actively refused it
```

tvanderbruggen, 2022-11-28:
Hi,

There is some library version incompatibility with Ubuntu 22.04 (see issue #542).
Could you try to install a Ubuntu 20.04 virtual machine and rebuild the instrument ?

bamiesking, 2022-11-28:
Hi Thomas, thanks for your reply. That seems to have fixed the issue completely.