Koheron/koheron-sdk #559: What is the unit of adc-dac-dma raw data?

hansmdau97, 2023-02-12:
Hi team,

I am trying to measure the NEP of a PD using alpha250. 

Referring to the site here (https://www.koheron.com/blog/2018/01/22/measuring-photodetector-noise-equivalent-power)

When using the adc-dac-dma example, I wonder what the unit of adc raw data is.

The equipment used is koheron DRV200, Koheron PD01, Alpha250

tvanderbruggen, 2023-02-13:
Hi,

The ADC data are not converted into a physical unit, they are the raw ADC data stored in a 16 bits integer.
Knowing the ADC input span `input_range` in Volts (typ. 1 Vpp on 50 Ohm), you can convert the raw data into Volts.
For example converting channel 0 data to Volts:

```py
data0_volts = driver.adc[0] * input_range / 2**16
```

The [Ltc2157 driver](https://github.com/Koheron/koheron-sdk/blob/master/boards/alpha250/drivers/ltc2157.hpp) provides the calibrated input range.
You can add this function to the class `AdcDacDma`

```py
  @command(classname="Ltc2157")
  def get_input_voltage_range(self, channel):
      return self.client.recv_float()
```

and then use it as

```py
data0_volts = driver.adc[0] * driver.get_input_voltage_range(0) / 2**16
```

You can check your code by sending a sine of known amplitude over 50 Ohm and compared to the DMA data rescaled with the `input_range` factor.

I hope this answers your question.

hansmdau97, 2023-03-23:
Your answer was very helpful thank you.

And I have another question. I am trying to find the voltage noise density using the data received using the adc-dac-dma instrument.

Is there a way to use the functions of the FFT instrument while using the adc-dac-dma instrument? KeyError: 'FFT' was output when the required function was added with "@command(classname = "FFT")"

tvanderbruggen, 2023-03-23:
No, only one instrument is running at the time, because it is not the same logic running on the FPGA.

But you can compute the FFT of the adc-dac-dma data using numpy on the python side.
See for example https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha15/adc-dac-dma/test_fft.py.