Koheron/koheron-sdk #571: ALPHA250 connection

Mohammed-678, 2023-06-27:
Hi I have a problem sending my bitstream file to the ALPHA250 card I configured  ethernet connection between the host and a Zynq board as precised on koheron website but I don't know what is the next step

tvanderbruggen, 2023-06-27:
I'm not sure I understand your issue.
When you enter the board IP address into a web browser, do you managed to access the instrument webpage ?

Mohammed-678, 2023-06-27:
No I don't

tvanderbruggen, 2023-06-27:
Did you try to set up a [direct ethernet connection](https://www.koheron.com/support/tutorials/setup-direct-ethernet-static-ip/) or do you connect the board to a local network ?

Mohammed-678, 2023-06-27:
I did a direct ethernet connection as specefied in the page that you sent

tvanderbruggen, 2023-06-27:
`10.42.0.100` seems like a good IP address for direct link, I guess you [assigned a static address](https://www.koheron.com/support/tutorials/assign-static-ip/) for the board ?

Problem must come from the configuration of your computer, do you use Windows or Linux ?

Mohammed-678, 2023-06-27:
I use windows

tvanderbruggen, 2023-06-28:
Alternate DNS server should be `10.42.0.255` instead of `10.42.0.2` (see https://www.koheron.com/support/tutorials/setup-direct-ethernet-static-ip/), but I'm not sure that's the issue.

Can you show the content of the `/etc/network/interfaces` file on the ALPHA250 board ? (Using
the [USB serial](https://www.koheron.com/support/tutorials/setup-usb-serial-connection/) )

Mohammed-678, 2023-06-28:
I have problems with assigning a static IP address to a Zynq board  I receive the error message "The disc image file is corrupted" when trying to mount a Koheron-alpha250_v0.22 file

tvanderbruggen, 2023-06-28:
The image is a Linux image, it will be difficult to mount it on Windows.

Did you try to connect to COM1 ? The USB serial port appears as a virtual COM port on Teraterm.

Mohammed-678, 2023-06-28:
I fixed the problem thank you very much it was a problrm recognizing the USB Serial Port , I just installed a VCP driver and then I did the same step and it worked

elhaouati, 2023-07-24:
Hi

 I'm facing nearly the same problem I didn't assign an IP adress to the board cause I can't edit the /etc/network/interfaces file, it says that it is on reading mode only (so it remains DHCP configuration) 
I configured my ip adress to 192.168.1.5
the led are indicating 255 
when I ping the 192.168.1.255 it can't find it

Kind regards