Koheron/koheron-sdk #575: ADC function

elhaouati, 2023-07-19:
Hi there,

I'm currently working on signal processing application, and I rely on the ADC-DAC-DMA example. I am particularly interested in the ADC and the samples it sends. For a signal coming from a voltage generator that is externally synchronized using the clock output of the board, at a frequency of 31.25 = 250/8 MHz, I expect it to send the same or approximately the same 8 samples every period on the adc0[15:0] output of the block adc_dac, and does adc0[15:2] respect the conversion table in page 22 of the [adc_datasheet:](https://www.analog.com/media/en/technical-documentation/data-sheets/21576514fb.pdf) 

Is my understanding of the ADC function correct ?

Thank you in advance.

jeanminet, 2023-07-19:
Hello,
Your understanding is correct.
The ADC in configured in two's complement format (https://github.com/Koheron/koheron-sdk/blob/master/boards/alpha250/drivers/ltc2157.hpp#L65). 
adc0[15:2] = `10 0000 0000 0000` corresponds to about -0.5 V (1 Vpp ADC range)
adc0[15:2] = `01 1111 1111 1111` corresponds to about +0.5 V

Here is the Python code that convert the two complement representation of adc0[15:0] in an integer between -2^15 and 2^15:
https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/adc-dac-dma/test.py#L46

elhaouati, 2023-07-20:
Hello,
Thank you for your response.
In the ADC-DAC-DMA example, the ADC sampling rate is 250 Msps, and the clock output of the ZYNQ-7 Processing System is at 143 MHz. This means that we are processing these samples at a frequency lower than the rate at which they arrive. Is this situation going to cause us some problems or limitations in our system.