Koheron/koheron-sdk #577: adc-bram example for ALPHA250-4

elhaouati, 2023-07-25:
Hi

I'm working on the adc-bram example, after sending my bitstream file to /dev/xdevcfg, I can't execute the python test file even after correcting the ip adress.
[image]

I did installed the requirements libraries such as koheron , pyyaml , jinja2 

Thanks in advance.

tvanderbruggen, 2023-07-25:
Hi,

To start an instrument please use the `make run` command.
See the [documentation](https://www.koheron.com/software-development-kit/documentation/).

elhaouati, 2023-07-25:
I used `$ make CONFIG=examples/alpha250-4/adc-bram/config.yml HOST=192.168.1.100 run` command to generate the project

tvanderbruggen, 2023-07-25:
Can you show me the instruments available by going to the 192.168.1.100 address on a web browser ?

elhaouati, 2023-07-25:
this what I've got
[image]

and after few seconds of the sending of my bitstream i can no longer reach this page

tvanderbruggen, 2023-07-25:
Can you tell me on which version of Ubuntu you are building the instrument ?

elhaouati, 2023-07-25:
I am using ubunto 20.04.1

tvanderbruggen, 2023-07-25:
That version should be fine.
Can you connect in ssh to the board and give the following logs:

* `journalctl -u koheron-server -n 200`
* `journalctl -u uwsgi -n 200`

elhaouati, 2023-07-25:
[image]
[image]

tvanderbruggen, 2023-07-25:
I don't understand why you use the command `cat /tmp/system_wrapper.bit > /dev/xdevcfg` ?
Loading of the bitstream is done when starting the instrument.

Can you send the bottom of `uwsgi` log after you callled `make run` for the `adc-bram` instrument ?

elhaouati, 2023-07-25:
I use the command `scp system_wrapper.bit root@192.168.1.100:/tmp` to send the bitstream file to the board and then `cat /tmp/system_wrapper.bit > /dev/xdevcfg` to load it to xdevcfg

I am not sure that I understand what do you mean with `make run` when and where should i use it

1. I used $ make CONFIG=examples/alpha250-4/adc-bram/config.yml HOST=192.168.1.100 run command to generate the project.
2.  I generated the bitsream file using vivado 
3. and then I sent it to the board and load it to xdevcfg

tvanderbruggen, 2023-07-25:
Running `make CONFIG=examples/alpha250-4/adc-bram/config.yml HOST=192.168.1.100 run` does the following:
- Generate project & bitstreams + build instrument server + package everything together
- Send the package to the HOST
- Start instrument: Load bitstream + server

So `make run` is all you have to do.
Now in your case, for some reason when you call it the `adc-ram` instrument is not loaded on the board...
Can you show me result of the call  `make CONFIG=examples/alpha250-4/adc-bram/config.yml HOST=192.168.1.100 run` ?

elhaouati, 2023-07-25:
[image]

elhaouati, 2023-07-25:
I did a `make clean all` command and it seemed that it has a problem 
[image]

So i deleted the folder in /adc-bram located in /tmp manually then I did a `make CONFIG=examples/alpha250-4/adc-bram/config.yml HOST=192.168.1.100 run` that was the result 
[image]

tvanderbruggen, 2023-07-25:
That's why it crashes.
What you can do is set internet access to your computer and then call `make CONFIG=examples/alpha250-4/adc-bram/config.yml` (no run) that will only build the instrument and fetch what is required from the net.

Once this is done you can reconnect the ethernet to the board and call `make CONFIG=examples/alpha250-4/adc-bram/config.yml HOST=192.168.1.100 run`, it will used what has already been downloaded.

elhaouati, 2023-07-25:
I did the `make CONFIG=examples/alpha250-4/adc-bram/config.yml` command and now it crashes at this step:
[image]

tvanderbruggen, 2023-07-25:
You need to install the Typescript copiler `tsc`.

Running `make setup_web` should do it.

elhaouati, 2023-07-25:
After installing tsc I've got this problem
[image]

elhaouati, 2023-07-25:
still the same SyntaxError message, is it a problem with the version of `tsc`

elhaouati, 2023-07-25:
I did some researches in some forums and I found that I can just change the syntax ?? with || and it worked then I did a `make run` command then I executed the python files after installing the required libraries such as Tkinter and PIL and it worked in the end for test.py

tvanderbruggen, 2023-07-25:
Good if it works !

elhaouati, 2023-07-25:
Thank you so much for your time and efforts