Koheron/koheron-sdk #579: Phase noise analyzer error when trying to build: No module named encodings, alpha 250

AravindAnthur, 2023-07-27:
I have ´VirtualBox installed with Ubuntu 18.04.6 LTS.
I installed Vivado 2020.1. I built the koheron-sdk. When I am trying to build the phase noise analyzer, it gives the following error as shown below.
I have unset the PYTHONPATH and PYTHONHOME in the main OS (Ubuntu). I am unable to do so in the tcl console. 
When I try to run python in the tcl console, I get the same error.
Why is it looking for python libraries in the Vivado when I have not set it? How can I unset when I am not able to run it in the TCL console?

--------------------------------------------Error---------------------------------------------------
Python path configuration:
  PYTHONHOME = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16'
  PYTHONPATH = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16:/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/bin:/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib:/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python2.7/lib-tk:/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python2.7/site-packages:'
  program name = 'python'
  isolated = 0
  environment = 1
  user site = 1
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python3.11'
  sys._base_executable = '/root/anaconda3/bin/python'
  sys.base_prefix = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16'
  sys.base_exec_prefix = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16'
  sys.platlibdir = 'lib'
  sys.executable = '/root/anaconda3/bin/python'
  sys.prefix = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16'
  sys.exec_prefix = '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16'
  sys.path = [
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/bin',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python2.7/lib-tk',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python2.7/site-packages',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/bin',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python311.zip',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python3.11',
    '/home/alpha250/Xilinx/Vivado/Vivado/2020.1/tps/lnx64/python-2.7.16/lib/python3.11/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'
-----------------------------------------------------------------------------------------------------------------------------

jeanminet, 2023-07-28:
I'm not able not reproduce the issue.
This look similar to this issue:
https://support.xilinx.com/s/article/72570?language=en_US

If the proposed solution does not work, a workaround is to run the Python script `fir.py` outside Vivado:
```
$ python3 fir.py 6 4 1 print
-1.618954e-05, 2.584474e-07, 3.812441e-05, 3.246789e-05, -4.482759e-05, -9.397516e-05, 9.083408e-07, 1.533381e-04, 1.167431e-04, -1.458971e-04, -2.844709e-04, 1.658415e-06, 4.107315e-04, 2.991348e-04, -3.554285e-04, -6.697500e-04, 1.512526e-06, 9.031381e-04, 6.420885e-04, -7.370271e-04, -1.361883e-03, -1.938097e-06, 1.754189e-03, 1.229428e-03, -1.373285e-03, -2.508251e-03, -1.362991e-05, 3.124516e-03, 2.172294e-03, -2.370201e-03, -4.303218e-03, -4.313074e-05, 5.227280e-03, 3.623665e-03, -3.869355e-03, -7.018622e-03, -1.093110e-04, 8.369834e-03, 5.816484e-03, -6.081126e-03, -1.108562e-02, -2.527117e-04, 1.306720e-02, 9.169187e-03, -9.376193e-03, -1.732767e-02, -5.754459e-04, 2.037691e-02, 1.461571e-02, -1.456318e-02, -2.773241e-02, -1.409761e-03, 3.309531e-02, 2.493035e-02, -2.393532e-02, -4.892995e-02, -4.441580e-03, 6.203012e-02, 5.358319e-02, -4.711181e-02, -1.229836e-01, -3.489931e-02, 2.081611e-01, 4.230792e-01, 4.230792e-01, 2.081611e-01, -3.489931e-02, -1.229836e-01, -4.711181e-02, 5.358319e-02, 6.203012e-02, -4.441580e-03, -4.892995e-02, -2.393532e-02, 2.493035e-02, 3.309531e-02, -1.409761e-03, -2.773241e-02, -1.456318e-02, 1.461571e-02, 2.037691e-02, -5.754459e-04, -1.732767e-02, -9.376193e-03, 9.169187e-03, 1.306720e-02, -2.527117e-04, -1.108562e-02, -6.081126e-03, 5.816484e-03, 8.369834e-03, -1.093110e-04, -7.018622e-03, -3.869355e-03, 3.623665e-03, 5.227280e-03, -4.313074e-05, -4.303218e-03, -2.370201e-03, 2.172294e-03, 3.124516e-03, -1.362991e-05, -2.508251e-03, -1.373285e-03, 1.229428e-03, 1.754189e-03, -1.938097e-06, -1.361883e-03, -7.370271e-04, 6.420885e-04, 9.031381e-04, 1.512526e-06, -6.697500e-04, -3.554285e-04, 2.991348e-04, 4.107315e-04, 1.658415e-06, -2.844709e-04, -1.458971e-04, 1.167431e-04, 1.533381e-04, 9.083408e-07, -9.397516e-05, -4.482759e-05, 3.246789e-05, 3.812441e-05, 2.584474e-07, -1.618954e-05

```

And replace the line in block_design.tcl
```
set fir_coeffs [exec $python $project_path/fir.py $n_stages $dec_rate_min $diff_delay print]
```
with the obtained FIR coefficients:
```
set fir_coeffs "-1.618954e-05, 2.584474e-07, 3.812441e-05, 3.246789e-05, -4.482759e-05, -9.397516e-05, 9.083408e-07, 1.533381e-04, 1.167431e-04, -1.458971e-04, -2.844709e-04, 1.658415e-06, 4.107315e-04, 2.991348e-04, -3.554285e-04, -6.697500e-04, 1.512526e-06, 9.031381e-04, 6.420885e-04, -7.370271e-04, -1.361883e-03, -1.938097e-06, 1.754189e-03, 1.229428e-03, -1.373285e-03, -2.508251e-03, -1.362991e-05, 3.124516e-03, 2.172294e-03, -2.370201e-03, -4.303218e-03, -4.313074e-05, 5.227280e-03, 3.623665e-03, -3.869355e-03, -7.018622e-03, -1.093110e-04, 8.369834e-03, 5.816484e-03, -6.081126e-03, -1.108562e-02, -2.527117e-04, 1.306720e-02, 9.169187e-03, -9.376193e-03, -1.732767e-02, -5.754459e-04, 2.037691e-02, 1.461571e-02, -1.456318e-02, -2.773241e-02, -1.409761e-03, 3.309531e-02, 2.493035e-02, -2.393532e-02, -4.892995e-02, -4.441580e-03, 6.203012e-02, 5.358319e-02, -4.711181e-02, -1.229836e-01, -3.489931e-02, 2.081611e-01, 4.230792e-01, 4.230792e-01, 2.081611e-01, -3.489931e-02, -1.229836e-01, -4.711181e-02, 5.358319e-02, 6.203012e-02, -4.441580e-03, -4.892995e-02, -2.393532e-02, 2.493035e-02, 3.309531e-02, -1.409761e-03, -2.773241e-02, -1.456318e-02, 1.461571e-02, 2.037691e-02, -5.754459e-04, -1.732767e-02, -9.376193e-03, 9.169187e-03, 1.306720e-02, -2.527117e-04, -1.108562e-02, -6.081126e-03, 5.816484e-03, 8.369834e-03, -1.093110e-04, -7.018622e-03, -3.869355e-03, 3.623665e-03, 5.227280e-03, -4.313074e-05, -4.303218e-03, -2.370201e-03, 2.172294e-03, 3.124516e-03, -1.362991e-05, -2.508251e-03, -1.373285e-03, 1.229428e-03, 1.754189e-03, -1.938097e-06, -1.361883e-03, -7.370271e-04, 6.420885e-04, 9.031381e-04, 1.512526e-06, -6.697500e-04, -3.554285e-04, 2.991348e-04, 4.107315e-04, 1.658415e-06, -2.844709e-04, -1.458971e-04, 1.167431e-04, 1.533381e-04, 9.083408e-07, -9.397516e-05, -4.482759e-05, 3.246789e-05, 3.812441e-05, 2.584474e-07, -1.618954e-05"

```

jeanminet, 2023-08-03:
I got the same error running Vivado 2020.1 on Ubuntu 22.04.
I don't have the error running Vivado 2017.2 on Ubuntu 22.04.