Koheron/koheron-sdk #584: Converting the DAC output voltage of the Alpha250 board

hansmdau97, 2023-11-02:
Hi,there

I'm trying to modulate a laser using an alpha250 and a DRV200, and I have a few questions about this.

1. when using the ADC-DAC-DMA (https://github.com/koheron/koheron-sdk/blob/master/examples/alpha250/adc-dac-dma/test.py), is it possible to adjust the output voltage of the DAC?

2. when using the DRV200 to modulate a laser, is it correct to use the ADC-DAC-DMA's driver.dac = (the waveform to be converted) and the set_dac() function to modulate the laser?

The laser I am using is an LP1550-SAD2.

tvanderbruggen, 2023-11-02:
At the lines 65-66 of the test.py script you have an example of how to set the DAC (here generating a frequency chirp):

```py
driver.dac = 0.9 * np.cos(2*np.pi * (fmin + chirp * t) * t)
driver.set_dac()
```

As you said, you first generate and fill the `driver.dac` buffer with the desired waveform and then call `driver.set_dac()` when you want to output the signal.

The waveform in the `driver.dac` buffer takes values between -1 (minimum DAC output = -0.5 V on 50 Ohm) and +1 (maximum DAC output = +0.5 V on 50 Ohm).

It is thus possible to adjust the output voltage by setting the waveform amplitude for example by replacing the 0.9 factor with 0.5: 

```py
driver.dac = 0.5 * np.cos(2*np.pi * (fmin + chirp * t) * t)
driver.set_dac()
```

Note that you can also use the modulation gain jumper of the DRV200 to modify the modulation amplitude of the laser current.

hansmdau97, 2023-11-06:
Thank you for your answer.
I understand the current modulation of the DAC, but when I measured using the adc-dac-dma example, the default voltage output of ADC0 was measured to be 0.11~0.12V.
Does this offset affect the measured data? Also, when writing code to remove this, is it correct to subtract the mean of the DMA measured data from the ADC data I want to measure when there is no input data?
`
driver.start_dma()
 driver.get_adc()
 driver.stop_dma()
offset = np.mean(driver.adc)
`
The data I want to measure after
`
driver.start_dma()
driver.get_adc()
driver.stop_dma()
driver.adc -= offset
`
Am I understanding this correctly?

hansmdau97, 2023-11-06:
Thank you for your answer.
I understand the current modulation of the DAC, but when I measured using the adc-dac-dma example, the default voltage output of ADC0 was measured to be 0.11~0.12V.
Does this offset affect the measured data? Also, when writing code to remove this, is it correct to subtract the mean of the DMA measured data from the ADC data I want to measure when there is no input data?
`driver.start_dma()`
`driver.get_adc()`
`driver.stop_dma()`
`offset = np.mean(driver.adc)`

The data I want to measure after
`driver.start_dma()`
`driver.get_adc()`
`driver.stop_dma()`
`driver.adc -= offset`

Am I understanding this correctly?