Koheron/koheron-sdk #587: Clock switching + tuning between LMK04906 input ref clks for ALPHA250

a-panella, 2024-01-16:
Hi Koheron support,

I am attempting to switch the reference clock for TCXO into the LMK04906 for the ALPHA250 I have.
I have written a python class that calls into ClockGenerator.hpp to do the following:

- change adc/dac sampling frequency to 200 MHz
- switch reference clock to TCXO manually `set_reference_clock(2)`
- tune the TCXO by calling `set_tcxo_clock(0xXX)`

I can confirm that the koheron-server.service is logging that the functions are being being called. Calling set_reference_clock DOES change the clock that is being used according to the logs. But the VCO frequency does not appear to change when trimming the DAC--> VCO path.

Additionally, measuring CLKO, the output 10MHz ref from the board, it appears to not change when TCXO is selected 

Can I please confirm that I haven't missed anything in implementing TCXO freq tuning? Thanks.

I note that in your init() function, you call `configure` after changing TCXO tuning, but I tried this and found that there is no change, as it looks like `configure(...)` is configuring the PLL dividers etc via SPI, but the DAC-->VCO is via I2C-0 . 

Essentially my goal is I want to be able to change the TXCO frequency by small amounts to account for drift. Is this possible? From the docs, it seems possible. 

Thanks

a-panella, 2024-01-17:
Additional information, even when setting reference clock to external, the ADC/DAC sampling clocks still work. So now I am thinking that TCXO tuning is fine, and setting reference may not be working correctly

tvanderbruggen, 2024-01-17:
Which version of the SDK are you using ?

The 0.23 fixes a bug in the PLL loop gains where sometimes the PLL won't lock to the reference.
In that case the PLL is deriving the clock from the free running VCXO.
This might be what you are observing.

Note that you have to use an image with the 0.23 installed, and not only an instrument build with the 0.23,
because PLL settings are done once at startup from the default instrument.

a-panella, 2024-01-21:
Thanks @tvanderbruggen after building 20.04 version of OS (0.23), I found I have control over the TCXO tuning

closed 2024-01-21 by a-panella