Koheron/koheron-sdk #588: Connecting Koheron OS to internet

kth7316, 2024-02-01:
Hello, 
I'm working on an Alpha board with Koheron OS, and i would like to retrieve in live data from internet, but i struggle to share internet connection of my pc to the Alpha250...
How can I do this ? Or is it possible to use an alternative way to have internet access ? 
Thanks !

tvanderbruggen, 2024-02-01:
Hi,

To setup a direct Ethernet connection between your PC and the board you may follow this user-guide:

https://www.koheron.com/support/tutorials/setup-direct-ethernet-static-ip/

kth7316, 2024-02-01:
Hello @tvanderbruggen and thanks for your answer, 

I tried your configuration but without success...
I am using a Windows as host but if you have a Linux host required solution I can move on. On my host I shared the internet connection from the ethernet3 to ethernet5 (not indicated in your guide is it normal ?)

Here is my network diagram is it ok according to you ?

[image]

I connect to the alpha via SSH from my PC: [192.168.1.100:22](https://192.168.1.100:22/)
The problem is when I disable my proxy on my Alpha i can curl the [192.168.1.100](https://192.168.1.100:22/) webpage. But for other internet adresses I can't connect at all. And when i set the proxy i can curl nothing ...

One of the main problem is I think that I have a proxy.

It says :

```
curl: (6) Could not resolve host: www.google.com
or curl: (6) Could not resolve host: http://proxy.me
```

Here is my conf:

In /etc/profile and /etc/bash.bashrc, I set : [http://proxy.me/](http://proxy.me/%60)

In : /etc/resolv.conf i set :
```

nameserver 12.40.4.88
nameserver 12.1.1.108

```

In /etc/network/interfaces i set :

```
source-directory /etc/network/interfaces.d
allow-hotplug eth0
# Static IP
iface eth0 inet static
address 192.168.1.100
gateway 12.40.3.91
netmask 255.255.255.0
network 192.168.1.0
broadcast 192.168.1.255

```

Thanks

tvanderbruggen, 2024-02-01:
Well usually if you setup a direct connection you can only communicate to the board and not the internet.

The easiest way to have both accesses is to connect the ALPHA and your computer to a router...

kth7316, 2024-02-01:
Ok thanks i will try that ! 
Thank you again M. @tvanderbruggen

closed 2024-02-20 by kth7316