Koheron/koheron-sdk #589: Password not accepted for ssh

ndm1019, 2024-02-23:
Hello,

I am working with a red pitaya and using the "Red Pitaya v0.22" disk image for Zynq provided here:
[https://www.koheron.com/software-development-kit/documentation/ubuntu-zynq/](url)

Where I simply downloaded the image and used the Disk Image Writer to format it onto an SD card as outlined here:
[https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html#linux](url)

When I ping the RP there seems to be no issue, but when I try to SSH the provided password (changeme) fails to authenticate. Is there a step I'm missing between formatting the SD and being able to interface via SSH? 

For added context, I have my RP and computer connected via ethernet cable to a 24-port netgear switch box (intranet) for them to communicate. The error I get when attempting ssh is: 'Permission Denied (publickey, password)'.

Any help would be greatly appreciated!

vanhoesens, 2024-02-26:
I am having the exact same problem.
Any ideas??

AdrikVab, 2024-02-27:
Are you sure your ssh connection is reaching the Red Pitaya? You should be able to run one of the example designs (i.e. LED Blinker) without having to ssh into the rp. Does that work?

ndm1019, 2024-02-27:
I'm fairly certain it's reaching the rp because I disconnected everything else from my computer and was still able to ping it. The ssh attempt with everything disconnected gave me the same issue.

When I try to run the LED blinker it starts processing in the terminal, eventually throws the error:
make: *** [web/web.mk:22: tmp/examples/zedboard/led-blinker//web/app.js] Error 1

not sure what this error is indicating, any thoughts?

ndm1019, 2024-02-28:
I believe I found my issues here:

1) when playing with some network files on my computer it turns out I accidentally assigned the RP's IP address to my computer. So I was actually just pinging myself and not the RP
2) Following the steps here:
[https://www.koheron.com/support/tutorials/assign-static-ip/](url)
corrected my network settings on the RP to make it identifiable on my network.
3) I am using a red pitaya and was running the LED blinker protocol in the zedboard library, which obviously is not compatible.

after following these steps I am able to ping and ssh to my RP as expected.

closed 2024-02-28 by ndm1019