Koheron/koheron-sdk #590: Test.py not working

ndm1019, 2024-02-28:
Hello,

I am working with a red pitaya and using the "Red Pitaya v0.22" disk image for Zynq provided here:
[https://www.koheron.com/software-development-kit/documentation/ubuntu-zynq/](https://github.com/Koheron/koheron-sdk/issues/url)

Where I simply downloaded the image and used the Disk Image Writer to format it onto an SD card as outlined here:
[https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html#linux](https://github.com/Koheron/koheron-sdk/issues/url)

I have installed anaconda and created an environment where I'm trying to run koheron's python files. The version I have is:
 conda version : 24.1.2
 conda-build version : 24.1.2
 python version : 3.11.7.final.0

When in the koheron-sdk directory in my conda environment, I try to run the python test file with:
python test.py

which gives the following output:

Traceback (most recent call last):
  File "/home/me/anaconda3/envs/Koheron_env/lib/python3.12/site-packages/koheron/koheron.py", line 248, in __init__
    self.sock.connect((host, port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/me/Documents/koheron-sdk/examples/red-pitaya/led-blinker/test.py", line 10, in <module>
    client = connect(host, name='led-blinker')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/me/anaconda3/envs/Koheron_env/lib/python3.12/site-packages/koheron/koheron.py", line 58, in connect
    client = KoheronClient(host)
             ^^^^^^^^^^^^^^^^^^^
  File "/home/me/anaconda3/envs/Koheron_env/lib/python3.12/site-packages/koheron/koheron.py", line 251, in __init__
    raise ConnectionError('Failed to connect to {}:{} : {}'.format(host, port, e))
requests.exceptions.ConnectionError: Failed to connect to 192.168.1.100:36000 : [Errno 111] Connection refused

Any help on fixing this issue would be greatly appreciated!

AdrikVab, 2024-02-29:
A similar issue was reported and resolved last year: https://github.com/Koheron/koheron-sdk/issues/556

It was resolved by installing an Ubuntu 20.04 virtual machine. 

Does this help your situation?

ndm1019, 2024-03-16:
Hello, 

this resolved my issue. Thank you!

closed 2024-03-16 by ndm1019