Koheron/koheron-sdk #591: Speed up compilation/synthesis/implementation process

kth7316, 2024-03-04:
Hello, 

My team uses Alpha250 and RedPitaya cards with koheron SDK. 
We really appreciate your product. 

However, the compilation/synthesis/implementation time is very long for a complex TCL design.

I'd like to explore possibilities to speed up the process. 

One of the main problems is that when we modify a verilog or vhdl file, and run a execute command, I get :

```
source ./fpga/vivado/core.tcl
...
# file delete -force $output_path/$core_name $output_path/$project_name.cache $output_path/$project_name.hw $output_path/$project_name.xpr
# create_project -part $part $project_name $output_path
ERROR: [Common 17-53] User Exception: Project already exists on disk, please use '-force' option to overwrite: 
     /home/pc355/koheron-sdk/tmp/examples/alpha250/dmpp/cores/dmp.ip_user_files
INFO: [Common 17-206] Exiting Vivado at Mon Mar  4 16:24:46 2024...
fpga/fpga.mk:31: recipe for target 'tmp/alpha250/dmpp/cores/dmp/component.xml' failed
Error 1

```
So I have to clean up before starting again. I lose everything that was previously compiled and I have to do it again for a single file. 
Do you have a solution to this problem? Is it possible to cache certain cores, modules and compile only the modified cores? 
We're also planning to use Ccache for C++ drivers.

Secondly, it seems that the implementation phase is the bottleneck. As it depends on Vivado, do you have any advice? 
I've seen some stuff on the internet but haven't tried it yet. Also, do you have any hardware specifications that you recommend?

pavel-demin, 2024-03-04:
Try replacing the `file delete ...` line in `core.tcl` with the following:
```
file delete -force $output_path/$core_name $output_path/$core_name.cache $output_path/$core_name.hw $output_path/$core_name.ip_user_files $output_path/$core_name.sim $output_path/$core_name.xpr
```
I recently updated this line to remove all files and directories created by recent versions of Vivado.

jeanminet, 2024-03-04:
I just did `$ time make CONFIG=examples/alpha250/fft/config.yml`

```
....
[tmp/examples/alpha250/fft//fft.zip] OK

real	9m1,428s
user	7m54,485s
sys	0m18,150s

```
I have an Intel® Core™ i9-9900K CPU, with SSD and 64 GB RAM running Ubuntu 22.04.4 LTS.
The 64 GB RAM is probably overkill...

kth7316, 2024-03-04:
Ok thanks a lot, 
I will try that today.

So @pavel-demin the line that you send enables to not recompile everythings ?

pavel-demin, 2024-03-04:
> So @pavel-demin the line that you send enables to not recompile everythings ?

Yes, the updated line fixes the error from your first message and you won't need to rebuild all IP cores every time you change the code of a single IP core.

I see that Jean has already updated the `core.tcl` script in this repository.

pavel-demin, 2024-03-05:
I have never used checkpoints myself, but I do not see why they would not work. The Tcl scripts create normal Vivado projects and all Vivado functionality should be available to them.

closed 2024-03-12 by kth7316