Koheron/koheron-sdk #593: Never receive the response of a command (or with huge delay)

jefedufeu, 2024-03-28:
Hello, 

I've created a program in PyQt and at certain times ( ~20 - 40 min) it freezes when I retrieve the contents of a buffer via the command: `data = self.client.recv_vector(dtype='uint32')`

The response takes a long time to arrive, so my program freezes. How can I overcome this kind of problem? 

I've thought of doing it in multithreading so as not to be blocked, but if I send two python commands at the same time to the koheron OS, how will it handle it?  I could implement a queue that requests and sends them one after the other. 

A try except block with TimeoutException? (from signal.SIGALRM for example) 

Do you have any ideas? especially as I don't have any information in thectl log.

Thanks,

tvanderbruggen, 2024-03-28:
Hi,

If I understand properly, you don't see an error message when calling `journalctl -koheron-server -u 200` on the board ?

Do you see something when running `dmesg` ?

Which version of the OS are you using ?
It could be the same issue as https://github.com/Koheron/koheron-sdk/issues/528.

The issue might also be on the client side.

Regarding multithreading, you can open several connections with the server, so you can for example have one socket open per thread. There is a lock at the level of each driver, so that only one request from one connection can be executed at a time on a given driver, but you can call simultaneously different drivers. However I don't think this will fix your problem ...

kth7316, 2024-04-08:
Mr. @tvanderbruggen,

You say : 
> you can open several connections with the server, so you can for example have one socket open per thread

I try this in a python file by setting a common TCP port

`client = custom_connect(host, 35359, env.INSTRUMENT_NAME, restart="true")`

But i got error : `requests.exceptions.ConnectionError: Failed to connect to 192.168.1.103:35359: [WinError 10061] No connection could be established because the target computer expressly refused it`

After a fast check I see that that the tcp_port is set in the server conf file : `server/core/config.hpp`

I will try to edit some files to have the possibility to have multiple socket per thread. But with you answer I understand that there is maybe an easier solution.

Thanks, 

@jefedufeu if you have some solution I'm also interested.

tvanderbruggen, 2024-04-08:
I don't know what your `custom_connect` does nor what you're trying to achieve with the TCP port ?

Something like this should work:

```py
from koheron import connect

client1 = connect(host, 'my_instrument', restart=True)
client2 = connect(host, 'my_instrument', restart=False)
```

Now if you call sequentially `client1` and `client2` you don't really benefit from having 2 connections,
but it might be beneficial to open one client per thread.

Obviously, you don't want to restart the instrument when opening the 2nd connection since it will close the first one.

tvanderbruggen, 2024-04-08:
`restart = True` closes the currently running instrument (in particular all the connections) are start it again.
In practice it kills the server process and start it over.
This acts as a reset so that the instrument is in a known state after the connect call.

Each client returned by `connect` opens its own socket, so if you are sending commands in parallel via different clients you are using different sockets.
On the server side a separated thread is open for each socket, so the server can receive commands in parallel.
As I said earlier, there is a lock a each driver so if you send in parallel commands to the **same driver** they will effectively  be queued (albeit there is no guarantee of execution order).
Commands send in parallel to **different drivers** can be executed in parallel by the server.

chrisjbillington, 2024-12-13:
@jefedufeu I wonder if you are seeing this issue https://github.com/Koheron/koheron-sdk/issues/569

If so, the workaround that worked for us was to install `ethtool` on the ALPHA250 and add `post-up ethtool -K eth0 sg off` to the network configuration in `/etc/network/interfaces`