Koheron/koheron-sdk #599: GLIBC incompatibility cross-compiling with Ubuntu 22.04

tvanderbruggen, 2024-04-18:
For reference here is the reason for the incompatibility of GLIBC versions when cross-compiling on Ubuntu 22.04 and deploying on an image using Ubuntu 20.04 or 16.04 

```
root@koheron:~# objdump -T /tmp/live-instrument/serverd | fgrep GLIBC_2.3
00000000  w   DF *UND*	00000000  GLIBC_2.34  pthread_join
00000000  w   DF *UND*	00000000  GLIBC_2.34  __pthread_key_create
00000000  w   DF *UND*	00000000  GLIBC_2.34  pthread_once
00000000      DF *UND*	00000000  GLIBC_2.34  pthread_create
00000000      DF *UND*	00000000  GLIBC_2.33  fstat64
00000000  w   DF *UND*	00000000  GLIBC_2.34  pthread_detach
00000000      DF *UND*	00000000  GLIBC_2.34  __libc_start_main
```

The incompatible symbols are related to pthread:
https://developers.redhat.com/articles/2021/12/17/why-glibc-234-removed-libpthread#the_developer_view

rsarwar87, 2024-04-18:
Is this from the docker build?

tvanderbruggen, 2024-04-18:
No it works fine with the Docker build.

I just wanted to document the root cause of the issue when building in native.