Koheron/koheron-sdk #611: Changing sampling frequency on my Board Alpha250-4

aida152, 2024-09-06:
Hello, I am currently working on a phase detection program with the Alpha Koheron 250-4. I have made some modifications to the program already available on GitHub with the adc-bram. My sampling frequency is 250 MHz, and I am able to perform quadrature phase detection when I input a 62.5 MHz signal. Now, I would like to change my sampling frequency to 100 MHz to perform phase detection with a 25 MHz input signal. I only changed the adc_clk in the FPGA code because I use this clock for my phase detection blocks, but when I test this new code, it still works for 62.5 MHz and not for 25 MHz, as if my modifications were not taken into account. Are there any other configurations I need to make? Thank you for your help.

tvanderbruggen, 2024-09-06:
Hi,

The ADC sampling rate is defined by the [clock generator](https://github.com/Koheron/koheron-sdk/blob/master/boards/alpha250-4/drivers/clock-generator.hpp) driver.
Use the `set_sampling_frequency()` function to select a given configuration defined in `clock_cfg`.

I've just added a 100 MHz config for the ALPHA250-4 (see https://github.com/Koheron/koheron-sdk/pull/612).
The 100 MHz config index is 3, so calling `set_sampling_frequency(3)` will set the sampling frequency to 100 MHz.

NB: If you add your own clock configuration don't forget to set the MMCM phase-shift using the script [test_phase_shift.py](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250-4/adc-bram/test_phase_shift.py).
See this issue https://github.com/Koheron/koheron-sdk/issues/526.

aida152, 2024-09-06:
Thank you for your response, but where should I call the set_sampling_frequency function? 
Here is my code to get the phase values,
#!/usr/bin/env python
# -*- coding: utf-8 -*-

import numpy as np
import os
import time
from adc_bram import AdcBram
from koheron import connect
from test_ import frac

import matplotlib
matplotlib.use('TKAgg')
from matplotlib import pyplot as plt

def signed_binary(num, num_bits):
    if num >= 0:
        binary_str = format(num, f'0{num_bits}b')  # Convert to unsigned binary
    else:
        binary_str = format((1 << num_bits) + num, f'0{num_bits}b')  # Convert to two's complement
        
    return binary_str

#if j>0 else fichier.write(str(j+360) + "\n")
host = os.getenv('HOST', '192.168.1.100')
client = connect(host, 'adc-bram', restart=False)
driver = AdcBram(client)
fichier= open('PM.txt','w')

for i in range(1): # 1 SEULE ACQUISITION

    driver.trigger_acquisition() #déclence acquisition

    driver.get_adc(0) 
    data=driver.get_adc(1) 
    	
    a=(np.int32(data >> 16) - 32768) % 65536 - 32768 #récupérer 16 bits de 	poids fort
    #a=(np.int32(data % 65536) - 32768) % 65536 - 32768 #recupèrer 16 bits de poids faible

    b=[signed_binary(e,16) for e in a] 
    resultat =[((-frac(d)*180/np.pi)) for d in b]

    [fichier.write(str(j) + "\n") for j in resultat]

fichier.close()

Should I call it here?

tvanderbruggen, 2024-09-06:
You can call it after initializing the driver:

```py
driver = AdcBram(client)
driver.set_sampling_frequency(3)
```

aida152, 2024-09-06:
I found where to call it, and it works well now. I added these five lines to my code:
```
clk_200MHz = {'idx': 0, 'fs': 200E6}
clk_250MHz = {'idx': 1, 'fs': 250E6}
clk_100MHz = {'idx': 2, 'fs': 100E6}

clock = clk_100MHz
driver.set_sampling_frequency(clock['idx'])
```
and it works when I select the clock frequency. Thank you for helping me

aida152, 2024-09-06:
I have another question. What if I want to synchronize my board with another clock using the CLKIN input? Thank you for your response.

tvanderbruggen, 2024-09-06:
Place a 10 MHz reference at the CLKIN and add
```py
 driver.set_reference_clock(0) # External
```
to your script

aida152, 2024-09-09:
Thank you for your help :)

closed 2024-09-11 by jeanminet