Koheron/koheron-sdk #622: Some issues encountered while trying to use the phase noise analyzer

AlSnAs-Sean, 2025-01-11:
Hello, 
I am now trying to measure the phase noise of a TSL570 laser produced by santec using Alpha250. Last time, I asked some questions about the (n, fs) in the python script, and I really appreciate your answer. However, I encountered some new issues. My work bases mainly on the website 'Measuring the phase noise of a laser/Koheron'. 

First, in the python script I found cic_rate=20, I wonder whether the cic_rate can be changed or not. There's a function called set_cic_rate in the python examples provided on the github website, is it possible to set the cicrate in the Alpha250 by using this function? 

Second, the psd results from the python script from the 'Measuring the phase noise of a laser' website is quite large 1e21, I think there might be some mistakes in the script. Maybe we need to add '/len(data)' to the code 'psd += 2.0 * np.abs(np.fft.rfft(window * data)) ** 2' and get 'psd += 2.0 * np.abs(np.fft.rfft(window * data)/len(data)) ** 2', so that we can get a correct fft result? And why do we need to divide the psd by fs * np.sum(window ** 2) in the code 'psd /= (fs * np.sum(window ** 2)) # rad^2/Hz'? To be honest, I am not certain how to caculate the psd in python, and I do not know how to match the psd therotically caculated with the psd in the python script provided. 

Third, there is a calib_factor 4.196 and a const 8192 when we try to change gradians to radians, could you tell me what do they indicate? Please forgive me for my poor ability in English expression.
Thanks! I really appreciate your answer.

tvanderbruggen, 2025-01-11:
Hi,

On the ALPHA250 phase noise analyzer design the CIC rate is adjustable between the limits:

* cic_decimation_rate_min: 4
* cic_decimation_rate_max: 8192

The actual decimation rate can be set using the [set_cic_rate(uint32_t rate)](https://github.com/Koheron/koheron-sdk/blob/19b092683ceec13ce6122ef7663a8292034cdd8b/examples/alpha250/phase-noise-analyzer/phase-noise-analyzer.hpp#L64) function.

The script in the blog page plots the SSB phase noise in dBc/Hz.
You can divide it by f^2 to obtain the frequency noise.

When performing the FFT you obtain the power spectrum (in rad^2), you further need to divide by the sampling frequency fs to obtain the power spectral density (in rad^2/Hz).
The factor `np.sum(window ** 2)` compensates for the effective bandwidth of the FFT window.

The FPGA core extracting the signal phase introduces some difficult to predict scaling factors.
Those have been calibrated using a signal generator to obtain a signal in radians at the output.

I hope this answers your questions.

AlSnAs-Sean, 2025-01-14:
Thank you for your answer! It really helps.
However, I try to measure the phase noise of a moku:go instrument. I generated a sin wave (10Vpp,20MHz) by the moku and sent the output directly to a Alpha250. But the result seems weird. I accessed rawdata of phase in three different cicrate [5,10,20], and I ploted the three caculated SSB phase noise in one figure. However, the three lines do not match with each other properly. Is it reasonable that they do not match?
Here is the result.
[image]

Thank you for your help!

tvanderbruggen, 2025-01-14:
Do you compensate for the change in sampling frequency when you compute the phase noise ?

You can also look at the result for [PhaseNoiseAnalyzer::get_phase_noise()](https://github.com/Koheron/koheron-sdk/blob/19b092683ceec13ce6122ef7663a8292034cdd8b/examples/alpha250/phase-noise-analyzer/phase-noise-analyzer.hpp#L196) to compare ...

jeanminet, 2025-01-14:
> I generated a sin wave (10Vpp,20MHz) by the moku and sent the output directly to a Alpha250

The ALPHA250 ADC input  is only 1Vpp.

AlSnAs-Sean, 2025-01-17:
> The ALPHA250 ADC input range is only 1Vpp.

Thank you. I have adjusted the amplitude to 1Vpp.

> Do you compensate for the change in sampling frequency when you compute the phase noise 

I am not sure what does 'compensate' indicate here, do you mean modifying the frequency range? 
fs = 200E6 / (cic_rate * 2)
n_avg = 100
f = np.arange((n/2 + 1)) * fs / n
These are the code from the website script, can they compensate for the sampling rate?

Otherwise, I try to compare the result of phase noise from the website and the python script, but they seem a little different. The noise floor is similar, but the 1/f noise is different. So I am wondering how do the website analysis the raw data? Does it use the same as the python script on the website 'Measuring the phase noise of a laser | Koheron'?

The two results are displayed here.

[image]

[image]

tvanderbruggen, 2025-01-17:
The FFT power spectrum is scaled the same way to obtain the PSD in the python script:

```py
psd /= n_avg
psd /= (fs * np.sum(window ** 2)) # rad^2/Hz
```

and in the [app](https://github.com/Koheron/koheron-sdk/blob/19b092683ceec13ce6122ef7663a8292034cdd8b/examples/alpha250/phase-noise-analyzer/phase-noise-analyzer.hpp#L219):

```c++
phase_noise[i] /= (fs * window.W2() * fft_navg); // rad^2/Hz
```

Which FFT window are you using ? The web app uses an Hann window...

AlSnAs-Sean, 2025-01-17:
> Which FFT window are you using ? The web app uses an Hann window

I used a square window,  `window = np.ones(n)`, maybe that makes sense. I will change the window and compare again.
Could you tell me where can I see the code of the app?

tvanderbruggen, 2025-01-17:
[Here is the code of the app](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/phase-noise-analyzer/phase-noise-analyzer.hpp)

AlSnAs-Sean, 2025-02-24:
Thank you for your answer! It really helps.
However, I have a new question. When I plot the phase noise psd, there are always some peaks which I think should not appear. For example, in the plot below, there are about 8 peaks between 1kHz to 10kHz. Do you know the reason?
Thank you for your attention.

[image]

jeanminet, 2025-02-25:
The peak at about 2 kHz and its harmonics might be caused by the fan.
Can you try disconnecting the fan when you take the measurement?

lkmzx, 2025-11-25:
> The peak at about 2 kHz and its harmonics might be caused by the fan. Can you try disconnecting the fan when you take the measurement?

Are there any commands to control the fan's on and off? I want to automatically turn off the fan during testing and then automatically enable it when finished.