Koheron/koheron-sdk #623: Problems setting the num_average_min in the python environment

bastizaj, 2025-01-28:
Hi, 
I want to use the python environment to change the num_average_min without the web app running. If the web app is running at the same time, I can control it with python commands. But without the web app I get unusually high returns for set_num_avergae_min, e.g. 184787. Is there a solution for this problem? 

Regards

jeanminet, 2025-01-29:
Do you use the `spectrum` or the `oscillo` example ?:
https://github.com/Koheron/koheron-sdk/tree/master/examples/red-pitaya/oscillo
https://github.com/Koheron/koheron-sdk/tree/master/examples/red-pitaya/spectrum

bastizaj, 2025-01-29:
I used the spectrum_analyzer.py from laser-development-kit as a starting point, but it should be similar. 
I added the driver.set_num_average_min(xxx) command, but without the web app running, it did not accept the "xxx" value and printed/used some strange high values as described above.