Koheron/koheron-sdk #627: Estimating PSD from the ADC data not matching FFT noise floor estimate from Alpha-250

AravindAnthur, 2025-04-04:
I am trying to estimate the power spectral density from the ADC data obtained from the class AdcDacDma given here - https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/adc-dac-dma/test.py. This is so that I can measure noise upto 100 Hz of frequencies. I have terminated all ports of alpha-250 acquisition board with 50 Ohm terminators.
I write additional function as follows to estimate the psd, following the phase noise measurement portion (https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/phase-noise-analyzer/phase_noise_analyzer.py). However, the estimate that I get from the function given below does not match what I get from https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/fft/python/test_noise_floor.py. Is there any additional processing of adc data required to get correct PSD?

#############################Code################################################
def psd(self, navg, window='hann', verbose=False):
        numpts = 8192
        win = signal.get_window(window, Nx=numpts)
        f = np.arange((numpts // 2 + 1)) * self.fs / numpts
        psd = np.zeros(f.size)

        power = 0
        
        driver.start_dma()
        driver.get_adc()
        
        print(np.shape(driver.adc[:numpts]))
        
        for i in range(navg):
            if verbose:
                print("Acquiring sample {}/{}".format(i + 1, navg))
            
            datareadadc = driver.adc[:numpts]
            psd += np.abs(np.fft.rfft(win * datareadadc)) ** 2
           
        driver.stop_dma()
        print(power / navg)

        psd /= navg
        psd /= (self.fs * np.sum(win ** 2)) # rad^2/Hz

        return f, psd

###############################End of code###################################

tvanderbruggen, 2025-04-06:
Hi,
I think you're missing a factor 2:

psd += 2.0 * np.abs(np.fft.rfft(win * datareadadc)) ** 2

instead of:

psd += np.abs(np.fft.rfft(win * datareadadc)) ** 2

AravindAnthur, 2025-04-08:
Hi, 
Thanks for your response. That is still not solving the issue. 
I get about 1e-2 V/sqrt(Hz) at 100 kHz with the psd estimation from the ADC data. But 3e-7 V/sqrt(Hz) at 100 kHz (this is approximately the number I am expecting I believe, for 50 Ohm terminated ADC input) using the FFT noise floor code.
Thanks.