Koheron/koheron-sdk #628: Questions Regarding DAQ Buffer Size and SDK Compatibility with Alpha250-4

yeojunpark, 2025-04-21:
Hello,

I am currently using the Alpha250-4 from Koheron.

I would like to change the size of the DAQ buffer—or the number of data points acquired per acquisition. It is currently set to 256 MB, but I would like to reduce it to 4 MB. How can I modify the buffer size or acquisition settings?

Additionally, I am interested in using the Phase Noise Analyzer for Alpha250 with the Alpha250-4.
Is this SDK compatible with the Alpha250-4? If so, could you please let me know how to set it up with this device? Are any configuration changes required?

Thank you very much for your support.

tvanderbruggen, 2025-04-22:
Hi,

I suppose you are using the adc-bram instrument.
In the `memory` section of the [config.yml](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250-4/adc-bram/config.yml) you can edit the adc memory ranges:

```yml
  - name: adc0
    range: 256K
    offset: '0x40000000'
  - name: adc1
    range: 256K
    offset: '0x40040000'
```

Using '4K' instead of '256K' should work (I suppose you meant 256k instraed of 256M).

Regarding the phase noise analyzer you can copy the project, and modify a few links in the [config.yml](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/phase-noise-analyzer/config.yml).

```yml
board: boards/alpha250-4

...

xdc:
  - boards/alpha250-4/config/ports.xdc

drivers:
  - boards/alpha250-4/drivers/common.hpp
  - boards/alpha250-4/drivers/ltc2157.hpp
  - boards/alpha250-4/drivers/clock-generator.hpp
  - boards/alpha250/drivers/eeprom.hpp
  - boards/alpha250/drivers/gpio-expander.hpp
  - boards/alpha250/drivers/temperature-sensor.hpp
  - boards/alpha250/drivers/power-monitor.hpp
  - boards/alpha250/drivers/precision-adc.hpp
  - boards/alpha250/drivers/precision-dac.hpp
  - boards/alpha250/drivers/spi-config.hpp
  - server/drivers/dma-s2mm.hpp
  - ./dds.hpp
  - ./phase-noise-analyzer.hpp
```

You also need to remove any connection to the RF DAC, mostly disconnect the DDS from the DAC output (see [line 33 of the BD](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/phase-noise-analyzer/block_design.tcl#L33))

yeojunpark, 2025-05-13:
Thank you for your response.

I have a couple of follow-up questions:

1. When you refer to “256K,” do you mean 256 kilosamples?
2. Is it possible to continue acquiring analog data while the buffer is being cleared? In other words, can data acquisition and transmission to the host computer occur simultaneously?

Thank you very much for your assistance.

tvanderbruggen, 2025-05-14:
Hi,

256K = 256 x 1024 = 262144 bits, this corresponds to 8192 samples of 32 bits.
Each ADC channel sample is on 16 bits (two extra dummy bits on top of the actual 14 bits) then 2 ADC channels are concatenated to get a 32 bits sample.

The BRAM that stores the data is configured as a "True Dual-Port Memory" so you can write data on one port while reading from the other port.

There is no effect on the jitter, the ADC sampling clock is provided by the clock generator and runs independently of the FPGA acquisition.