Koheron/koheron-sdk #634: ALPHA15 adc-dac-dma example missing points

AveryWong03, 2025-07-09:
When running the adc-dac-dma example for the ALPHA15 the first 77 data points are consistently mismatched, corresponding to a previous acquisition. 

I believe this issue arises due to the signal adc_dac/adc_valid being sent to the axi-stream (axis_dwidth_converter/s_axis_tvalid) prior to the axi-dma being started/setup. To fix this, a module can be made to only assert adc_valid once the axi-dma has asserted tready (axi_dma/s_axis_s2mm_tready). 

With this solution, there are still 8 sample points that appear to be apart of a previous acquisition. This may be explained by the [AXI DMA LogiCORE IP Product Guide (PG021)](https://docs.amd.com/r/en-US/pg021_axi_dma/Typical-System-Interconnect)

> In the absence of any setup (that is, before it is programmed to run), AXI DMA will pull the s_axis_s2mm_tready signal Low after taking in four beats of streaming data. This will throttle the input data stream. To have a minimum amount of throttling, ensure that the AXI DMA is set up to run much before the actual data arrives.

Since the stream is 64-bits, made up of 2 32-bit sample points (18-bit padded), I assume that the axi-dma will take 4 64-bit samples, giving us the 8 32-bit sample points. A more complete solution could involve an enable control signal for adc_valid that is asserted by the python application long after the axi-dma is started.

jeanminet, 2025-07-09:
Thanks !
I did not have time to test it on the board but I think something along these lines could work:
https://github.com/Koheron/koheron-sdk/commit/b4f251d004bf9ae49f

AveryWong03, 2025-07-10:
Hi Jean, thanks for the quick response and fix. 

I tested out these changes and found that your fix has solved the 8 sample point issue corresponding to the four beats of streaming data, but does not fix the issue with the other 69 sample points. I think this is because tvalid is still being sent down the stream prior to the dma starting, (to the dwidth converter, clock converter and tlast geneartor) causing data to fill up in the stream. So when the dma starts, those first 69 points (taken at some prior time) are transferred and only then is the expected data collected and transferred through. 

I've implemented a fix that I have tested on the board using a module that gates tvalid by an enable signal that is apart of the ps_ctl registers. That fix is implemented in [this](https://github.com/Koheron/koheron-sdk/pull/635) pull request. No pressure to merge the pull request, I just wanted to give an idea as to how a fix may be implemented.